koide3/direct_visual_lidar_calibration #131: Dynamic integrator behavior with horizontal movement of LiDAR

VeeranjaneyuluToka, 2024-11-22:
Hi,

I noticed that dynamic integrator works fine when there is yaw angle/vertical LiDAR movement but it does not work correctly when there is pitch angle / horizontal LiDAR movement while capturing data. 

Is there any limitation in dynamic integrator wrt pitch angle movement?

koide3, 2024-11-24:
Hmm, basically there is no specific assumptions on the sensor motion in the dynamic integrator, and it should work for pitch and horizontal movements. If possible, could you share a video of the processing result?

VeeranjaneyuluToka, 2024-12-11:
Here is the prerocessing output when i have movement horizontally. You can clearly notice that the vertical lines are not fused correctly .
/home/veeru/Pictures/Screenshots/Screenshot from 2024-12-11 10-14-06.png