koide3/direct_visual_lidar_calibration #135: insertion of deskewed points into ivox discards normals

themightyoarfish, 2024-12-06:
I am trying to fuse pointclouds with normals with the integrated_ct_icp_factor, and it crashes when trying to access the frame normals for computing the error. I tracked this down to this code in the dynamic integrator

```
  auto deskewed = std::make_shared<FrameCPU>(factor->deskewed_source_points(values));
  deskewed->add_covs(covariance_estimation.estimate(deskewed->points_storage, neighbors));
  deskewed->add_intensities(points->intensities, points->size());
  target_ivox->insert(*deskewed);
```

Here, the ivox structure complains about this frame not having normals, whereas the first inserted frame did have normals. So it appears the the ct_icp factor is currently broken. Or am I misunderstanding something?

koide3, 2024-12-07:
`deskewed` has only covariance matrices and lacks normals. You need to add normals by calling `add_normals` if you want to use CT_ICP. There is also CT_GICP that uses covariances matrices, and it should work with `deskewed` in your post.

BTW, we released [gtsam_points](https://github.com/koide3/gtsam_points) that includes refined implementations of iVox and CT_ICP, which may be more suitable to your purpose.

themightyoarfish, 2024-12-07:
How would I go about adding the normals there? My input points to the dynamic integrator already do have normals.

koide3, 2024-12-07:
You can let `covariance_estimation` estimate normals in addition to covs:

```cpp
std::vector<Eigen::Vector4d, Eigen::aligned_allocator<Eigen::Vector4d>> normals;
std::vector<Eigen::Matrix4d, Eigen::aligned_allocator<Eigen::Matrix4d>> covs;
covariance_estimation.estimate(deskewed->points_storage, neighbors, normals, covs);

auto deskewed = std::make_shared<FrameCPU>(factor->deskewed_source_points(values));
deskewed->add_normals(normals);
deskewed->add_covs(covs);
deskewed->add_intensities(points->intensities, points->size());
target_ivox->insert(*deskewed);
```

It is also possible to transform normals of your input point cloud using the interpolation result of CT_ICP. This way will be more efficient but complicated. If you want to try it, use `get_time_table()`, `get_time_indices()`, and `get_source_poses()` in CT_ICP.

https://github.com/koide3/direct_visual_lidar_calibration/blob/105ca482e7ebfb4efcb4ee07662987a9470b9374/include/vlcal/common/integrated_ct_icp_factor.hpp#L64