koide3/direct_visual_lidar_calibration #141: Error opening file: /camera/image during preprocessing in direct_visual_lidar_calibration

BreCaspian, 2025-02-19:
### **Issue Title:**
`Error opening file: /camera/image` during preprocessing in `direct_visual_lidar_calibration`

### **Description:**

I am facing a recurring issue while using the `direct_visual_lidar_calibration` package to preprocess ROSbag files. The tool fails during the preprocessing step, and throws the following error:

```
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: /camera/image
Aborted (core dumped)
```

more

```
yao@RM:~/catkin_ws$ rosrun direct_visual_lidar_calibration preprocess /media/yao/RM/livox_ros1 /media/yao/RM/rosbag_preprocessed -a --camera_model plumb_bob --camera_intrinsic 1452.0,1453.1,0.2897,0.2058 --camera_distortion_coeffs 0.1392,-1.6770,0.0,0.0,0.0
data_path: /media/yao/RM/livox_ros1
dst_path : /media/yao/RM/rosbag_preprocessed
input_bags:
- /media/yao/RM/livox_ros1/rosbag2_2023_03_09-13_42_46.bag
- /media/yao/RM/livox_ros1/rosbag2_2023_03_09-13_44_10.bag
- /media/yao/RM/livox_ros1/rosbag2_2023_03_09-13_44_54.bag
- /media/yao/RM/livox_ros1/rosbag2_2023_03_09-13_46_10.bag
- /media/yao/RM/livox_ros1/rosbag2_2023_03_09-13_46_54.bag
topics in /media/yao/RM/livox_ros1/rosbag2_2023_03_09-13_42_46.bag:
- /livox/points : sensor_msgs/PointCloud2
- /livox/imu : sensor_msgs/Imu
- /image : sensor_msgs/Image
- /camera_info : sensor_msgs/CameraInfo
selected topics:
- camera_info: /camera_info
- image      : /image
- points     : /livox/points
intensity_channel: intensity
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: /image
Aborted (core dumped)

```

This issue occurs consistently when trying to preprocess ROSbag files containing the `/camera/image` or `/image` topic. The error prevents me from continuing with the calibration process.

### **Steps to Reproduce:**

1. Run the following preprocessing command for the first ROSbag:
   ```bash
   rosrun direct_visual_lidar_calibration preprocess /media/yao/RM/rosbag_test /media/yao/RM/rosbag_preprocessed -a --camera_model plumb_bob --camera_intrinsic 1452.0,1453.1,0.2897,0.2058 --camera_distortion_coeffs 0.1392,-1.6770,0.0,0.0,0.0
   ```

2. The tool tries to process the ROSbag `lidar_camera_data_test1.bag` and its topics:
   - `/camera/camera_info` (sensor_msgs/CameraInfo)
   - `/camera/image` (sensor_msgs/Image)
   - `/cloudpoints` (sensor_msgs/PointCloud2)

3. The tool fails and returns the following error:
   ```
   terminate called after throwing an instance of 'rosbag::BagIOException'
   what():  Error opening file: /camera/image
   Aborted (core dumped)
   ```

4. I also ran the same preprocessing command for another set of ROSbags:
   ```bash
   rosrun direct_visual_lidar_calibration preprocess /media/yao/RM/livox_ros1 /media/yao/RM/rosbag_preprocessed -a --camera_model plumb_bob --camera_intrinsic 1452.0,1453.1,0.2897,0.2058 --camera_distortion_coeffs 0.1392,-1.6770,0.0,0.0,0.0
   ```

5. The same error occurs while processing the `rosbag2_2023_03_09-13_42_46.bag` in the `livox_ros1` directory.

### **Expected Behavior:**
The preprocessing step should process the `/camera/image` or `/image` topic and other selected topics successfully, without errors, and output the preprocessed data to the destination directory (`/media/yao/RM/rosbag_preprocessed`).

### **Actual Behavior:**
The preprocessing fails with the error `Error opening file: /camera/image`. The tool is unable to read or process the image data from the `/camera/image` or `/image` topic in the ROSbag files.

### **Troubleshooting Steps Taken:**

1. **Verified the availability and content of `/camera/image` and `/image` topics**:
   - I ran `rostopic echo /camera/image` and confirmed that messages are being published.
   - I played the ROSbag files and viewed the image data using `rqt_image_view` to ensure the image data is valid and visible.

2. **Checked the ROSbag file contents using `rosbag info`**:
   - I confirmed the presence of `/camera/image` or `/image` topics in the ROSbags, and checked their message count:
     ```bash
     rosbag info /media/yao/RM/rosbag_test/lidar_camera_data_test1.bag
     ```
     The output showed that the `/camera/image` topic has 93 messages, and `/cloudpoints` has 97 messages.

3. **Re-indexed the ROSbag files**:
   - I attempted to re-index the ROSbag file using:
     ```bash
     rosbag reindex /media/yao/RM/rosbag_test/lidar_camera_data_test1.bag
     ```

4. **Checked file permissions**:
   - I ensured that the file permissions are set correctly:
     ```bash
     sudo chmod -R 755 /media/yao/RM/rosbag_test
     ```

5. **Tried processing the files individually**:
   - I attempted to process only the `lidar_camera_data_test1.bag` file with no success:
     ```bash
     rosrun direct_visual_lidar_calibration preprocess /media/yao/RM/rosbag_test/lidar_camera_data_test1.bag /media/yao/RM/rosbag_preprocessed -a --camera_model plumb_bob --camera_intrinsic 1452.0,1453.1,0.2897,0.2058 --camera_distortion_coeffs 0.1392,-1.6770,0.0,0.0,0.0
     ```

6. **Checked for correct topic paths and message types**:
   - I verified that the topic paths and message types are correctly specified and match those in the ROSbag:
     ```bash
     rostopic type /camera/image
     ```
     The output confirmed that `/camera/image` has the correct `sensor_msgs/Image` type.

### **System Information:**
- **OS**: Ubuntu 20.04
- **ROS Version**: ROS Noetic
- **OpenCV Version**: 4.10.0

### **Logs:**
Error log:
```
terminate called after throwing an instance of 'rosbag::BagIOException'
what():  Error opening file: /camera/image
Aborted (core dumped)
```

### **Questions:**
1. Could there be an issue with how the /camera/image or /image topic is being serialized or read by the direct_visual_lidar_calibration tool?
2. Is there a potential mismatch between the image encoding format in the ROSbag and the tool's expectations?
3. Could the issue be related to the OpenCV version installed? Could an incompatible OpenCV version be causing this issue?
4. Have you encountered this issue before?
5. What steps should I take to thoroughly resolve this issue? Are there any specific debug or logging options I should enable to gather more information about the cause?

---

Thank you very much !

huanghuzz, 2025-03-21:
我遇见了和你一样的问题，请问您是否解决了该问题

BreCaspian, 2025-03-23:
Hi, sorry for the late reply due to recent work commitments.

I’ve actually encountered the same issue as you described. Unfortunately, I haven’t found a direct solution either. After some investigation, I believe the root cause lies in the compatibility issues between the source code and the specific versions of ROS and OpenCV. As of now, this problem cannot be resolved without modifying the original source code.

I’ve already submitted a detailed issue regarding this to the original repository and am currently waiting for the maintainers to respond or push an update.

That said, there is a good workaround — using **Docker**. If you're familiar with Docker, you’ll know that it eliminates the need to manually configure complex local environments. In fact, this reflects one of Docker’s core advantages: **providing a consistent and reproducible environment across systems**.

Fortunately, the author has already provided an official **Docker-based solution** in the project’s documentation. You can simply follow those instructions to run the tool and get your calibration results without dealing with local environment issues.

Hope this helps!

Best regards.

BloodLemonS, 2025-04-25:
> 我已经提供了相关建议，希望你的问题已经解决。
> 
> 祝好！
> […](#)
> 
>  我遇见了和你一样的问题，请问您是否解决了该问题
> 
>  <[#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009)>,
>  com>
>  ([koide3/direct_visual_lidar_calibration#141](https://github.com/koide3/direct_visual_lidar_calibration/issues/141))
>  <[#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009)>
> 
>  我遇见了和你一样的问题，请问您是否解决了该问题
> 
>  <[#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009)>,
>  com>

您好，我用docker配置了环境，但是在使用过程中docker非常的卡顿， 我是在工控机上使用的，没有显卡和cuda，请问这样有影响吗？会导致使用过程中卡顿吗？但是我再docker外ubuntu环境下安装使用就没有问题

BreCaspian, 2025-04-26:
> > 我已经提供了相关建议，希望你的问题已经解决。
> > 祝好！
> > […](#)
> > 我遇到了和你一样的问题，请问您是否解决了该问题
> > 直接回复此邮件，在 GitHub 上查看
> > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >，
> > 或取消订阅
> > 。
> > 您收到此邮件是因为您订阅了此主题。消息
> > ID：_**@**_ .*** 
> > com> 
> > [image: huanghuzz] _huanghuzz_留下了一条评论
> > ( [koide3/direct_visual_lidar_calibration#141](https://github.com/koide3/direct_visual_lidar_calibration/issues/141) ) 
> > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >
> > 我遇到了和你一样的问题，请问您是否解决了该问题
> > 直接回复此邮件，在 GitHub 上查看
> > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >，
> > 或取消订阅
> > 。
> > 您收到此邮件是因为您订阅了此主题。消息
> > ID：_**@**_ .*** 
> > com>
> 
> 您好，我用docker配置了环境，但是在使用过程中docker非常的卡顿，我是在工控机上使用的，没有显卡和cuda，请问这样有影响吗？会导致使用过程中卡顿吗？但是我在docker外ubuntu环境下安装使用就没有问题

建议您直接在您的笔记本或者台式机上使用docker运行该标定工具，如果缺少GPU和CUDA会导致大量计算无法有效进行，工控机性能较差。

祝好！

BloodLemonS, 2025-04-27:
> > > 我已经提供了相关建议，希望你的问题已经解决。
> > > 祝好！
> > > […](#)
> > > 我遇到了和你一样的问题，请问您是否解决了该问题
> > > 直接回复此邮件，在 GitHub 上查看
> > > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >，
> > > 或取消订阅
> > > 。
> > > 您收到此邮件是因为您订阅了此主题。消息
> > > ID：_**@___ ._**
> > > com>
> > > [image: huanghuzz] _huanghuzz_留下了一条评论
> > > ( [koide3/direct_visual_lidar_calibration#141](https://github.com/koide3/direct_visual_lidar_calibration/issues/141) )
> > > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >
> > > 我遇到了和你一样的问题，请问您是否解决了该问题
> > > 直接回复此邮件，在 GitHub 上查看
> > > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >，
> > > 或取消订阅
> > > 。
> > > 您收到此邮件是因为您订阅了此主题。消息
> > > ID：_**@**_ .***
> > > com>
> > 
> > 
> > 您好，我用docker配置了环境，但是在使用过程中docker非常的卡顿，我是在工控机上使用的，没有显卡和cuda，请问这样有影响吗？会导致使用过程中卡顿吗？但是我在docker外ubuntu环境下安装使用就没有问题
> 
> 建议您直接在您的笔记本或者台式机上使用docker运行该标定工具，如果缺少GPU和CUDA会导致大量计算无法有效进行，工控机性能较差。
> 
> 祝好！

我在工控机不用docker配置环境的条件下能够顺利运行，但是用docker就不行了。在工控机的外置环境下也是没有cuda和显卡

BreCaspian, 2025-04-28:
> > > > 我已经提供了相关建议，希望你的问题已经解决。
> > > > 祝好！
> > > > […](#)
> > > > 我遇到了和你一样的问题，请问您是否解决了该问题
> > > > 直接回复此邮件，在 GitHub 上查看
> > > > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >，
> > > > 或取消订阅
> > > > 。
> > > > 您收到此邮件是因为您订阅了此主题。消息
> > > > ID：_**@___ ._**
> > > > com>
> > > > [image: huanghuzz] _huanghuzz_留下了一条评论
> > > > ( [koide3/direct_visual_lidar_calibration#141](https://github.com/koide3/direct_visual_lidar_calibration/issues/141) )
> > > > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >
> > > > 我遇到了和你一样的问题，请问您是否解决了该问题
> > > > 直接回复此邮件，在 GitHub 上查看
> > > > < [#141 (comment)](https://github.com/koide3/direct_visual_lidar_calibration/issues/141#issuecomment-2742121009) >，
> > > > 或取消订阅
> > > > 。
> > > > 您收到此邮件是因为您订阅了此主题。消息
> > > > ID：_**@**_ .***
> > > > com>
> > > 
> > > 
> > > 您好，我用docker配置了环境，但是在使用过程中docker非常的卡顿，我是在工控机上使用的，没有显卡和cuda，请问这样有影响吗？会导致使用过程中卡顿吗？但是我在docker外ubuntu环境下安装使用就没有问题
> > 
> > 
> > 建议您直接在您的笔记本或者台式机上使用docker运行该标定工具，如果缺少GPU和CUDA会导致大量计算无法有效进行，工控机性能较差。
> > 祝好！
> 
> 我在工控机不用docker配置环境的条件下能够顺利运行，但是用docker就不行了。在工控机的外置环境下也是没有cuda和显卡

如果能够运行工具完成标定过程那么则无须多虑，没有显卡和CUDA不知是否能够顺利运行，可能求解过程较慢。

期待您的实验与回复！